nativescript-community/ble #218: Ios give me long uuid while android give me macaddress style

Atomico001, 2021-05-04:
Hello, i have a problem with ios and android data.

in ios i have this:
{
"UUID" : 83621c1a-7cfd-3d3b-b223-46fecbc9ec8"
...
}
while on android i have:
{
"UUID": DC:0D:30:04:2F:C3
...
}

Why?

farfromrefug, 2021-05-04:
@Atomico001 you need to look at native API documentations for device ids. This is the normal behavior.

closed 2021-05-04 by farfromrefug

Atomico001, 2021-05-04:
Thanks for answer, but i download an ios app to scan beacons and i can see the mac address there.
Another thing is that that uuid 83621c1a-7cfd-3d3b-b223-46fecbc9ec8 is nowhere in my beacon config. Is it a fake uuid?

farfromrefug, 2021-05-04:
@Atomico001 that is what i am talking about this is not the mac address. this is a unique id given by the system. You should read the native docs