nativescript-community/ble #219: Build error - can't resolve p-queue

chuckmitchell, 2021-05-07:
Trying to build a project that build yesterday -- might be an NPM issue?
I tried to manually add p-queue package to no effect. :(

Package.json

`{
  "name": "@nativescript/template-hello-world-ng",
  "main": "main.js",
  "version": "7.0.9",
  "author": "NativeScript Team <oss@nativescript.org>",
  "description": "NativeScript Application",
  "license": "SEE LICENSE IN <your-license-filename>",
  "publishConfig": {
    "access": "public"
  },
  "keywords": [
    "nativescript",
    "mobile",
    "angular",
    "{N}",
    "template"
  ],
  "repository": "<fill-your-repository-here>",
  "bugs": {
    "url": "https://github.com/NativeScript/NativeScript/issues"
  },
  "dependencies": {
    "@angular/animations": "~11.0.0",
    "@angular/common": "~11.0.0",
    "@angular/compiler": "~11.0.0",
    "@angular/core": "~11.0.0",
    "@angular/forms": "~11.0.0",
    "@angular/platform-browser": "~11.0.0",
    "@angular/platform-browser-dynamic": "~11.0.0",
    "@angular/router": "~11.0.0",
    "@nativescript-community/ble": "^3.0.14",
    "@nativescript-community/ui-material-textfield": "^5.2.8",
    "@nativescript/angular": "~11.0.0",
    "@nativescript/core": "~7.1.0",
    "@nativescript/theme": "~3.0.0",
    "p-queue": "^7.1.0",
    "reflect-metadata": "~0.1.12",
    "rxjs": "^6.6.0",
    "zone.js": "~0.11.1"
  },
  "devDependencies": {
    "@angular/compiler-cli": "~11.0.0",
    "@nativescript/android": "7.0.1",
    "@nativescript/types": "~7.0.0",
    "@nativescript/webpack": "~4.0.0",
    "@ngtools/webpack": "~11.0.0",
    "typescript": "~4.0.0"
  },
  "private": "true",
  "readme": "NativeScript Application"
}
`

`🐚  ns build android
Preparing project...
Hash: e74776b5dd36659135c1
Version: webpack 4.44.2
Time: 10620ms
Built at: 05/07/2021 6:14:47 PM
 3 assets
Entrypoint bundle = runtime.js vendor.js bundle.js
[../$$_lazy_route_resource lazy recursive] ../$$_lazy_route_resource lazy namespace object 160 bytes {bundle} [built]
[./app.css] 2.03 KiB {bundle} [built]
[./app/app-routing.module.ts] 1.16 KiB {bundle} [built]
[./app/app.component.ts] 883 bytes {bundle} [built]
[./app/app.module.ts] 1.44 KiB {bundle} [built]
[./app/device/device.service.ts] 1.35 KiB {bundle} [built]
[./app/device/devices.component.ts] 12.9 KiB {bundle} [built]
[./app/utils/byte_array_converter.ts] 1.34 KiB {bundle} [built]
[./main.ts] 1.9 KiB {bundle} [built]
[~/package.json] external "~/package.json" 42 bytes {bundle} [optional] [built]
    + 550 hidden modules

WARNING in Zone.js does not support native async/await in ES2017+.
These blocks are not intercepted by zone.js and will not triggering change detection.
See: https://github.com/angular/zone.js/pull/1140 for more information.

ERROR in ../node_modules/@nativescript-community/ble/bluetooth.js
Module not found: Error: Can't resolve 'p-queue' in '/Users/charles/projects/kanayo/andtool/node_modules/@nativescript-community/ble'
 @ ../node_modules/@nativescript-community/ble/bluetooth.js 4:0-29 951:29-35
 @ ./app/device/devices.component.ts
 @ ./app/app.module.ts
 @ ./main.ts
Executing webpack failed with exit code 2.`

farfromrefug, 2021-05-08:
yes my bad need to make a new release. p-queue 7.x seems to not work.
you can add p-queue in your project and force it to 6.x that will fix it

gurvan-guss, 2021-05-12:
Had the same issue building an old App for Android running `@nativescript-community/ble@3.0.19`
Fixed for now by adding `p-queue@6.6.2` package in my project

farfromrefug, 2021-05-12:
@gurvan-guss @chuckmitchell sorry for that just released 3.0.20 which should fix it