nativescript-community/ble #221: Property 'UUID' does not exist on type 'ICharacteristic'

mm-psymbl, 2021-05-18:
Looks like "Characteristic" is declared but not exported. Any plans to update this in the repo?

farfromrefug, 2021-05-18:
@mm-psymbl first you could give more details. Second one kind of remark is `Any plans to update this in the repo` ? 
Does it look like it is not maintained? Please show some respect

mm-psymbl, 2021-05-18:
@farfromrefug apologies, but I didn't mean to indicate that someone should get on it, what i meant was that if understood where the problem lies, is that something that's objectively wrong that will require some contributors to update the repo, or could it be the case that i am missing some types or something, given that others don't seem to have this problem?

As far as the problem goes, basically the angular demo, has an interface called ICharacteristic that extends Characteristic, but for some reason, the demo fails because UUID is not part of ICharacteristic, which is due to the fact that in bluetooth.d.ts, Characteristic is declared, but not exported. 

Sorry for the misunderstanding in the question, I should have elaborated better.

farfromrefug, 2021-05-18:
@mm-psymbl thanks for explaining. And no issue here, just with github, like any community thing, it is difficult to have the feel / tone of a message. And i like things clear ;)

Thanks for explaining the issue. I get it and just published 3.0.21 which exports all it can

mm-psymbl, 2021-05-19:
@farfromrefug no worries, it's my bad. I have somewhat of a talent for being too rush in my communication 😃 
Thanks a bunch

closed 2021-05-19 by mm-psymbl