nativescript-community/ble #222: Not getting data from heart rate monitor

mm-psymbl, 2021-05-20:
Hi all,

I have some very simple code to try reading data from my Polar OH1 but i can't seem to get any data from it.

I first start scanning like this:

```
doStartScanning(serviceUUIDs?: string[], seconds?: number) {
        const that = this;
        bluetooth.startScanning({
            filters: [{ serviceUUID: '180d' }],
            seconds: 5,
            onDiscovered: function (peripheral) {
                that.connectToPeripheral(peripheral.UUID);
            }
        }).then(function () {
            console.log("scanning complete");
        }, function (err) {
            console.log("error while scanning: " + err);
        });
    }
```

and then connectToPeripheral does this:

```
connectToPeripheral(uuid: string) {
        bluetooth.connect({
            UUID: uuid,
            onConnected: function (peripheral) {
                console.log("Periperhal connected with UUID: " + peripheral.UUID);

                const heartRateService = peripheral.services.find(s => s.UUID === '180d');
                console.log(heartRateService);
                bluetooth.startNotifying({
                    peripheralUUID: peripheral.UUID,
                    serviceUUID: heartRateService.characteristics[0].serviceUUID,
                    characteristicUUID: heartRateService.characteristics[0].UUID,
                    onNotify: function (result) {
                        console.log("read: " + JSON.stringify(result));
                    }
                }).then(function () {
                    console.log("subscribed for notifications");
                });
            },
            onDisconnected: function (peripheral) {
                console.log("Periperhal disconnected with UUID: " + peripheral.UUID);
            }
        });
    }
```

The outpput is this:

`CONSOLE LOG: read: {"peripheralUUID":"6e604f53-c7f8-95bc-ff96-8ca0199a647c","serviceUUID":"180d","characteristicUUID":"2a37","ios":{},"value":{}}`

value is always empty. Am i doing something wrong?

farfromrefug, 2021-05-20:
@mm-psymbl you need to understand how console log works. If you read the typings you ll see that value is Int8Array. Console log wont print it that s it. Read the value in it it works.

mm-psymbl, 2021-05-21:
@farfromrefug I am not sure i understand but I did just try to do this (which i found from an old ble project i tried) 

```
onNotify: function (result) {
    const data = that.arrayBufferToString(result.value);
    console.log(data);
}|
```

and console log prints nothing `CONSOLE LOG:` 

```
arrayBufferToString(buffer) {
        var byteArray = new Uint8Array(buffer);
        var str = "",
            cc = 0,
            numBytes = 0;
        for (let i = 0, len = byteArray.length; i < len; ++i) {
            var v = byteArray[i];
            if (numBytes > 0) {
                //2 bit determining that this is a tailing byte + 6 bit of payload
                if ((cc & 192) === 192) {
                    //processing tailing-bytes
                    cc = (cc << 6) | (v & 63);
                } else {
                    throw new Error("this is no tailing-byte");
                }
            } else if (v < 128) {
                //single-byte
                numBytes = 1;
                cc = v;
            } else if (v < 192) {
                //these are tailing-bytes
                throw new Error("invalid byte, this is a tailing-byte")
            } else if (v < 224) {
                //3 bits of header + 5bits of payload
                numBytes = 2;
                cc = v & 31;
            } else if (v < 240) {
                //4 bits of header + 4bit of payload
                numBytes = 3;
                cc = v & 15;
            } else {
                //UTF-8 theoretically supports up to 8 bytes containing up to 42bit of payload
                //but JS can only handle 16bit.
                throw new Error("invalid encoding, value out of range")
            }

            if (--numBytes === 0) {
                str += String.fromCharCode(cc);
            }
        }
        if (numBytes) {
            throw new Error("the bytes don't sum up");
        }
        return str;
    }
```

mm-psymbl, 2021-05-21:
I see, thank you :) I had expected value showing something albeit wrong.

closed 2021-05-21 by mm-psymbl