nativescript-community/ble #225: Error when running ns prepare ios --release

sebj54, 2021-06-03:
Hi there!

I am actually trying to compile my app for release in order to upload it on TestFlight.

When I run `ns prepare ios --release`, I am getting this error:

```
ERROR in ./node_modules/@nativescript-community/ble/bluetooth.ios.js + 2 modules
Cannot get final name for export '__esModule' of ./node_modules/@nativescript-community/ble/bluetooth.ios.js
while generating the root export '__esModule' (used name: '__esModule')
Error: Cannot get final name for export '__esModule' of ./node_modules/@nativescript-community/ble/bluetooth.ios.js
while generating the root export '__esModule' (used name: '__esModule')
    at getFinalBinding ([...]/node_modules/webpack/lib/optimize/ConcatenatedModule.js:464:10)
    at getFinalName ([...]/node_modules/webpack/lib/optimize/ConcatenatedModule.js:520:18)
    at [...]/node_modules/webpack/lib/optimize/ConcatenatedModule.js:1384:24
    at ConcatenatedModule.codeGeneration ([...]/node_modules/webpack/lib/optimize/ConcatenatedModule.js:1432:7)
    at [...]/node_modules/webpack/lib/Compilation.js:2712:22
    at [...]/node_modules/webpack/lib/Cache.js:93:5
    at Hook.eval [as callAsync] (eval at create ([...]/node_modules/webpack/node_modules/tapable/lib/HookCodeFactory.js:33:10), <anonymous>:6:1)
    at Cache.get ([...]/node_modules/webpack/lib/Cache.js:75:18)
    at ItemCacheFacade.get ([...]/node_modules/webpack/lib/CacheFacade.js:117:15)
    at Compilation._codeGenerationModule ([...]/node_modules/webpack/lib/Compilation.js:2705:9)
    at [...]/node_modules/webpack/lib/Compilation.js:2631:10
    at arrayIterator ([...]/node_modules/neo-async/async.js:3467:9)
    at processTicksAndRejections (internal/process/task_queues.js:77:11)
```

Everything works fine if I remove the `--release` flag.

Feel free to ask me anything to test, thanks for your help!

farfromrefug, 2021-06-03:
@sebj54 Never seen that error. seems related to your webpack config. you should ask on discord or slack to see if anyone faced such an issue.

sebj54, 2021-06-03:
I doubt it :( it works great for Android (even with `--release`) and my webpack config is barely the original one (I just add some constants and a rule to load SVGs with raw-loader.

```javascript
const webpack = require('@nativescript/webpack')
const dotenv = require('dotenv')
const { resolve } = require('path')
const fs = require('fs')

function parseDotenvValues(env){
    // This function just parses a .env file and returns it as an object
}

module.exports = (env) => {
    const dotenvValues = parseDotenvValues(env)

    webpack.init(env)

    webpack.chainWebpack((config) => {
        config.plugin('DefinePlugin').tap((args) => {
            Object.assign(args[0], {
                TNS_ENV: JSON.stringify(env.production ? 'production' : 'development'),
                ...dotenvValues,
            })
            return args
        })

        config.module
            .rule('svg')
            .test(/\.svg$/)
            .use('raw-loader')
            .loader('raw-loader')
    })

    // Learn how to customize:
    // https://docs.nativescript.org/webpack

    return webpack.resolveConfig()
}
```

Would it be because this plugin uses `@nativescript/webpack@3.0.0` and not `@nativescript/webpack@5.0.0-beta.x`?

I will try to hit Discord or Slack, thanks for the suggestion :)

farfromrefug, 2021-06-03:
@sebj54 ok can you try 3.0.25 ? Not sure i never saw that error in any of my apps. But anyway hopefully it will be fixed

sebj54, 2021-06-04:
It now builds correctly with the latest version! Thanks for the really quick update :)

closed 2021-06-04 by sebj54