nativescript-community/ble #226: startNotifying sometimes fails and make device disconnect

sebj54, 2021-06-16:
Hi there!

I use this plugin in a NativeScript app to handle a Bluetooth communication between a physical device and a phone. Communications are basically several writes followed by reads until the end of the communication.

Most of time it works perfectly, but sometimes (I would say once in 10 or 20) it fails and the startNotifying promise does not resolve nor reject. Also, the device is disconnected.

I wanted to give you some error logs or something but there isn't any! Not even in the Android logcat… 

The problem is happening:

- only on Android (as far as I can tell)
- on several characteristics
- at any moment in the communication (sometimes the first `startNotifying`, sometimes the last…)

I also added some logs in the `bluetooth.android.js` file (in the installed plugin directly) and I did not find where the problem is. 

Maybe it is an exception which is caught somewhere? That would explain why I don't have logs for this error.

Maybe it is related to this issue: https://github.com/nativescript-community/ble/issues/64?

Feel free to make me test anything! Thanks for your help :)

farfromrefug, 2021-06-28:
@sebj54 cant really help you with this one. I would say it is an android stack issue but can t be sure of anything.
If the device disconnects you should see some Android logs in logcat (although usually really hard to read).

farfromrefug, 2021-07-04:
@sebj54 should be released in 3.0.26! Thanks again

closed 2021-08-04 by sebj54