nativescript-community/ble #227: Root isn't granting Permission

SwaXs, 2021-07-01:
Im trying to search for Bluetooth devices.

And everytime I try to search for devices with this code:

` doStartScanning(){

        {
            this.isLoading = true;
            // reset the array
            this.observablePeripheralArray.splice(0, this.observablePeripheralArray.length);
            this.bluetooth.startScanning(
              {
                filters: [{serviceUUID:'180d'}],   // pass an empty array to scan for all services
                seconds: 4,         // passing in seconds makes the plugin stop scanning after <seconds> seconds
                onDiscovered: function (peripheral) {
                  this.observablePeripheralArray.push(fromObject(peripheral));
                }
              }
            ).then(function() {
              this.isLoading = false;
            },
            function (err) {
                this.isLoading = false;
              alert({
                title: "Whoops!",
                message: err,
                okButtonText: "OK, got it"
              });
            });
          };
    }`

I get this Error:
`JS: Unhandled Promise rejection: Permission denied ; Zone: <root> ; Task: null ; Value: Permission denied undefined`

Bluetooth is enabled and the permission granted from the Phone

farfromrefug, 2021-07-01:
@SwaXs clearly you are missing some permissions.
As you dont even give information about the platform you are running on, i ll close that issue and let you investigate.
Open a new issue if you find a real bug. Thanks

closed 2021-07-01 by farfromrefug

vdbongard, 2021-07-01:
We had a similar issue on Android. After adding permissions to the AndroidManifest.xml it worked:

```
	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
	<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
```

(Not sure if all permissions are needed though)