nativescript-community/ble #228: Discover new services after being connected

vdbongard, 2021-07-01:
Hello, 

we have the situation that we have a bluetooth device that has two GATT services.
By default only one service is visible which is the authentication service.
After the successful authentication the second service is also visible.

The problem is that the second service can not be found with this library after a successful authentication e.g. a `read` does throw a `service_not_found` exception.

We don't have the possibility to reconnect to discover the second service because the authentication is resetted on a disconnect.

We tried to do a discover manually with the undocumented functions `discoverServices` and `discoverAll` but they do not find the second service either.

Should `discoverServices` find new services of an already connected device or what is the expected behaviour of this function?

Is there any other option we can use to discover new services of an already connected device?

As a workaround both services are now visible by default, but the second service only gives correct values after a successful authentication.

farfromrefug, 2021-07-02:
@vdbongard i understand your issue. I suppose the issue is only on android?
As you can see here you can access a service only if it has been discovered https://developer.android.com/reference/android/bluetooth/BluetoothGatt#getService(java.util.UUID)
I think this is your issue: https://stackoverflow.com/questions/22596951/how-to-programmatically-force-bluetooth-low-energy-service-discovery-on-android. I could try and add the `refreshDeviceCache` as an option to `discoverServices` ?

vdbongard, 2021-07-02:
> I suppose the issue is only on android?

Yes we have this problem on Android. We did not test it on iOS though.

> I could try and add the refreshDeviceCache as an option to discoverServices ?

That would be great if this could be added!

farfromrefug, 2021-07-04:
@vdbongard could you try 3.0.26 there is a new `clearCache` option for `discoverServices`. Try to set it to `true` when you try to discover avec the authentication

vdbongard, 2021-07-06:
@farfromrefug Thank you for adding this so fast! Very much appreciated. I will try it out soon.

vdbongard, 2021-07-09:
I have just tested out the new version of the library with the `clearCache` option for `discoverServices` but unfortunately it does not discover the second service.

farfromrefug, 2021-07-09:
@vdbongard could you confirm you enter that function https://github.com/nativescript-community/ble/blob/519338ca60dc3e483ab0bc95e4059606756cdd07/src/bluetooth.android.ts#L2412 ? and that it does not throw here https://github.com/nativescript-community/ble/blob/519338ca60dc3e483ab0bc95e4059606756cdd07/src/bluetooth.android.ts#L2421 ?
You can the js equivalent of those files in node_modules/@nativescript-community/ble

vdbongard, 2021-07-09:
I have set a breakpoint in this file and it actually does throw:
```
TypeError: java.lang.Class[0] is not a constructor
    at Bluetooth.refreshDeviceCache (file:///data/data/org.nativescript.projectname/files/app/vendor.js:97061:70)
    at Bluetooth.discoverServices (file:///data/data/org.nativescript.projectname/files/app/vendor.js:97091:18)
    at Bluetooth.descriptor.value (file:///data/data/org.nativescript.projectname/files/app/vendor.js:97519:31)
    at BluetoothService.<anonymous> (file:///data/data/org.nativescript.projectname/files/app/bundle.js:3157:54)
    at Generator.next (<anonymous>)
    at fulfilled (file:///data/data/org.nativescript.projectname/files/app/bundle.js:3038:58)
    at ZoneDelegate.invoke (file:///data/data/org.nativescript.projectname/files/app/vendor.js:147824:26)
    at Object.onInvoke (file:///data/data/org.nativescript.projectname/files/app/vendor.js:74214:33)
    at ZoneDelegate.invoke (file:///data/data/org.nativescript.projectname/files/app/vendor.js:147823:32)
    at Zone.run (file:///data/data/org.nativescript.projectname/files/app/vendor.js:147574:43)
    at file:///data/data/org.nativescript.projectname/files/app/vendor.js:148319:34
    at ZoneDelegate.invokeTask (file:///data/data/org.nativescript.projectname/files/app/vendor.js:147857:31)
    at Object.onInvokeTask (file:///data/data/org.nativescript.projectname/files/app/vendor.js:74201:33)
    at ZoneDelegate.invokeTask (file:///data/data/org.nativescript.projectname/files/app/vendor.js:147856:36)
    at Zone.runTask (file:///data/data/org.nativescript.projectname/files/app/vendor.js:147624:47)
```

farfromrefug, 2021-07-09:
@vdbongard thanks! I published a new version for another try ... 3.0.27
Could you try ? (i cant try myself right now :s)
BTW you can the error logged if any by using something like this
```ts
import { BleTraceCategory } from '@nativescript-community/ble/bluetooth.common';
import { Trace } from '@nativescript/core';
Trace.addCategories(BleTraceCategory);
Trace.enable();
```

vdbongard, 2021-07-12:
Wow it did work now! The function does not throw anymore and the second service was discovered after authentication. Thank you very much!

closed 2021-07-12 by farfromrefug