nativescript-community/ble #230: StartNotifying not receinving messages on iOS

alessandrozanola, 2021-07-07:
Hi there!
I'm working on an app that must communicate with a device via BLE using this plugin.

- NativeScript version: 8.0.2
- BLE plugin version: 3.0.26

I'm using read and write characteristics with no issues, but now I'm trying to receive notifications from the device unsuccessfully. I'm sure that the device works correctly because I tested  it with LightBlue iOS app.

This is the code I use:

```js
await this.bluetooth.startNotifying({
  peripheralUUID: this.peripheral.UUID,
  serviceUUID: serviceUuid,
  characteristicUUID: characteristicUuid,
  onNotify: (result) => {
    console.log("Notified");
  }
})
console.log("Listening for notifications")
```

"Listening for notifications" is printed correctly but the notify callback is never reached.

Thanks for help!

farfromrefug, 2021-07-07:
@alessandrozanola could you force to version 3.0.25 and tell me if it work? Might be a regression

alessandrozanola, 2021-07-07:
I've tested versions `3.0.25` and `3.0.24` but anything changed.
This are my dependencies:

```json
"dependencies": {
  "@nativescript-community/ble": "3.0.25",
  "@nativescript/core": "~8.0.0",
  "@nativescript/theme": "~2.3.0",
  "axios": "^0.21.1",
  "crypto-js": "^4.0.0",
  "js-base64": "^3.6.0",
  "nativescript-barcodescanner": "^4.0.1",
  "nativescript-vue": "^2.8.1",
  "net": "^1.0.2",
  "vue-class-component": "^7.2.6",
  "vue-property-decorator": "^9.1.2",
  "vuex": "^3.6.2"
}
```

farfromrefug, 2021-07-07:
@alessandrozanola then I am not sure what could be happening.  I need to test in my apps to see if I have the issue.

alessandrozanola, 2021-07-08:
By the way I've tested it even with Android without success (all other Bluetooth features works correctly). Next days I'll try to debug it deeply and I'll let you know.

farfromrefug, 2021-07-08:
@alessandrozanola that s important. If it happens on both mostly it is related to the code you use in N.
Are you sure the `serviceUuid`, `characteristicUuid` are good?

alessandrozanola, 2021-07-09:
Yes, `serviceUuid` and `characteristicUuid` are good, I've even tried to write them directly as strings and I've changed the `characteristicUuid`, but still no results. Do you have other ideas?

alessandrozanola, 2021-07-09:
Ok, I found the problem, it was server side.
I forgot to add the descriptor `Client Characteristic Configuration` to the characteristic to enable notifications on it.

Hope this could help someone.

@farfromrefug thanks for help!

closed 2021-07-09 by alessandrozanola