nativescript-community/ble #231: Android transport type for GATT causing failures on some devices

keerl, 2021-07-09:
I am encountering strange behaviors when using this plugin on Android across different devices. On most phones it works perfectly, and on some other phones I get failures right after (or during) connection (Samsung Galaxy s10e).

I have been able to track this down to the `connectGatt()` function and the problem has to do with selecting the `transport` param. Currently this option is hard coded to `TRANSPORT_LE` (https://github.com/nativescript-community/ble/blob/master/src/bluetooth.android.ts#L1721) where the other options are `TRANSPORT_AUTO` and `TRANSPORT_BREDR` (https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback,%20int)). 

I experimented with my Galaxy s10e and found that things only work when I set the transport to `TRANSPORT_BREDR` or `TRANSPORT_AUTO`.

Is there a reason that this is hard-coded for `TRANSPORT_LE` and not `TRANSPORT_AUTO`?

farfromrefug, 2021-07-09:
@keerl good catch! i think this has been like this since the beginning of time. Did not even know it existed!
Fixed in 3.0.28

keerl, 2021-07-09:
Perfect! Works great. Thanks for the quick patch @farfromrefug

closed 2021-07-09 by keerl