nativescript-community/ble #232: Discover and connect to beacons

ju4nj3, 2021-07-24:
Please, could someone give some light to read or discover the UUIDs beacon slots, minor or major values?
Dumped entire peripheral object onConnected event, shows another UUID different to the configured on the beacon.
Also, if I setup two slots on the same beacon with same UUID, same major, and different minor, can't connect to the beacon.

And, why the UUID to connect is the MAC?

[image]

Peripheral connected with UUID: ED:21:03:98:66:E4

{
    "UUID": "ED:21:03:98:66:E4",
    "name": "iBKS105",
    "state": "connected",
    "services": [
        {
            "UUID": "1800",
            "characteristics": [
                {
                    "serviceUUID": "1800",
                    "UUID": "2a00",
                    "name": "2a00",
                    "properties": {
                        "read": true,
                        "write": true,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                },
                {
                    "serviceUUID": "1800",
                    "UUID": "2a01",
                    "name": "2a01",
                    "properties": {
                        "read": true,
                        "write": false,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                },
                {
                    "serviceUUID": "1800",
                    "UUID": "2a04",
                    "name": "2a04",
                    "properties": {
                        "read": true,
                        "write": false,
                        "writeWithoutResponse": false,
                        "notify": false,
                        "indicate": false,
                        "broadcast": false,
                        "authenticatedSignedWrites": false,
                        "extendedProperties": false
                    },
                    "descriptors": [],
                    "permissions": null
                }
            ]
[... 551 lines cut ...]
        "serviceUUID": "ff00",
        "UUID": "ff08",
        "name": "ff08",
        "properties": {
            "read": true,
            "write": true,
            "writeWithoutResponse": false,
            "notify": false,
            "indicate": false,
            "broadcast": false,
            "authenticatedSignedWrites": false,
            "extendedProperties": false
        },
        "descriptors": [],
        "permissions": null
    }
]
}
],
"localName": "iBKS105",
"manufacturerId": 76,
"advertismentData": {
"scanRecord": {}
}
}

farfromrefug, 2021-07-24:
@ju4nj3 you have to read it yourself. See https://stackoverflow.com/questions/18906988/what-is-the-ibeacon-bluetooth-profile

ju4nj3, 2021-07-25:
@farfromrefug Hi, thanks for the documentation. I readed a lot about beacons before I posted this, but my question is simple: 
How can I discover beacons by UUID/minor/major instead of service using this plugin? Anyone?

farfromrefug, 2021-07-25:
@ju4nj3 you need to look at native docs. This plugin is just a wrapper around native APIs

ju4nj3, 2021-07-26:
Thank you very much @farfromrefug 
Finally I found a simple solution using this plugin [https://github.com/demetrio812/nativescript-ibeacon] which still works.

closed 2021-07-26 by ju4nj3

farfromrefug, 2021-07-26:
@ju4nj3 seeing the code it makes sense to be in a different plugin. It is very specific code. Thanks for sharing!