nativescript-community/ble #238: Android: isConnected() always true

matteonatale, 2021-11-29:
Hi there!
I'm working on a Nativescript app that uses this plugin to interact with ble devices. I've noticed that in Android devices after the first connection, the `isConnected` method always returns true. To make it work properly again I have to close and reopen my app (this is because a singleton `Bluetooth` object is created during app startup). I've also added a call for `clear` method after every ble connection is closed, but it does not work either.

Is there any fix for this problem? Thanks! :D

farfromrefug, 2021-11-30:
@matteonatale thanks for the report. Should be fixed in the new version

matteonatale, 2021-11-30:
Cool!
Do you know when the next version will be available?

farfromrefug, 2021-11-30:
@matteonatale already is

matteonatale, 2021-12-01:
@farfromrefug still not working. Am I missing something?

farfromrefug, 2021-12-01:
What version are you on ? Please share more details and a precise way on how.to reproduce your issue

matteonatale, 2021-12-01:
`"@nativescript-community/ble": "^3.0.30"`
`"@nativescript/core": "^8.1.5"`

To reproduce the issue: connect to a BLE peripheral and disconnect, then the value returned from `isConnected` method is always `true`. The value returned before the connection was `false`.

farfromrefug, 2021-12-01:
@matteonatale you are using ^ which does not tell me which version you are using. PLease give me the verison from the node_modules folder.

matteonatale, 2021-12-01:
I've seen in the `package.json` of the plugin (inside node_modules) that the version used was `3.0.29`. I removed the `^` and after running `yarn install` the version is `3.0.30`.

The `package.json`
```json {
  "_args": [
    [
      "@nativescript-community/ble@3.0.30",
      "..."
    ]
  ],
  "_from": "@nativescript-community/ble@3.0.30",
  "_id": "@nativescript-community/ble@3.0.30",
  "_inBundle": false,
  "_integrity": "sha512-1rP58wS6ncuQT1zdbn4n2MwNkQ2eS6wi47l+BJn2LmeyOGBgATwIJke0vhs6n6uJIPWArNhv3oW85ph0u0YU4Q==",
  "_location": "/@nativescript-community/ble",
  "_phantomChildren": {},
  "_requested": {
    "type": "version",
    "registry": true,
    "raw": "@nativescript-community/ble@3.0.30",
    "name": "@nativescript-community/ble",
    "escapedName": "@nativescript-community%2fble",
    "scope": "@nativescript-community",
    "rawSpec": "3.0.30",
    "saveSpec": null,
    "fetchSpec": "3.0.30"
  },
  "_requiredBy": [
    "/",
    "/..."
  ],
  "_resolved": "https://registry.npmjs.org/@nativescript-community/ble/-/ble-3.0.30.tgz",
  "_spec": "3.0.30",
  "_where": "...",
  "author": {
    "name": "Eddy Verbruggen",
    "email": "eddyverbruggen@gmail.com"
  },
  "bugs": {
    "url": "https://github.com/nativescript-community/ble/issues"
  },
  "contributors": [
    {
      "name": "Brad Martin",
      "email": "bmartin@nstudio.io",
      "url": "https://github.com/bradmartin"
    },
    {
      "name": "Martin Guillon",
      "email": "martin.guillon@akyasl.fr"
    },
    {
      "name": "Ryan Pendergast",
      "email": "ryan.pendergast@gmail.com",
      "url": "https://rynop.com"
    }
  ],
  "dependencies": {
    "@nativescript-community/observable": "^2.0.10",
    "make-error": "1.3.6",
    "p-queue": "~6.6.2"
  },
  "description": "Connect to and interact with Bluetooth LE peripherals.",
  "gitHead": "0644fb9b6ff8ac55569326928f54a6bd712ba0b4",
  "homepage": "https://nativescript-community.github.io/ble",
  "keywords": [
    "ecosystem:NativeScript",
    "NativeScript",
    "iOS",
    "Android",
    "BLE",
    "Bluetooth",
    "Bluetooth LE",
    "Bluetooth Smart",
    "Bluetooth Low Energy",
    "Angular"
  ],
  "license": "MIT",
  "main": "./bluetooth",
  "name": "@nativescript-community/ble",
  "nativescript": {
    "platforms": {
      "android": "3.0.0",
      "ios": "3.0.0"
    }
  },
  "repository": {
    "type": "git",
    "url": "git+https://github.com/nativescript-community/ble.git"
  },
  "sideEffects": false,
  "typings": "./bluetooth.d.ts",
  "version": "3.0.30"
}

farfromrefug, 2021-12-01:
@matteonatale thanks can you try 3.0.31?

matteonatale, 2021-12-01:
I've tested version `3.0.31` on Android and it works correctly. Should I test it also on iOS or can I assume that everything will still work as expected?

farfromrefug, 2021-12-01:
@matteonatale no need i did not change anything on iOS! can i close this issue?

matteonatale, 2021-12-01:
Cool! 
I'm closing this issue now. Thank you for your quick replies @farfromrefug :D.

closed 2021-12-01 by matteonatale