nativescript-community/ble #239: read error after upgrade

ray007, 2021-12-09:
After upgrading from `3.0.29` to `3.0.31`, reading characteristics from one of our sensors does not work anymore - at least on android:
```
[BluetoothError]:service_not_found, undefined, undefined, {"peripheralUUID":"EB:38:0B:BE:FD:D7","serviceUUID":"78a90000-26ce-4367-bfeb-7e3c1d217561","characteristicUUID":"78a90001-26ce-4367-bfeb-7e3c1d217561","timeout":1000}
```
Downgrading back to `3.0.29` makes things work again.

farfromrefug, 2021-12-09:
@ray007 can you test 3.0.32 and report if it fixes it?

ray007, 2021-12-09:
@farfromrefug yes, version `3.0.32` seems to work too, thank you.

farfromrefug, 2021-12-09:
@ray007 cool can we close this issue then?

ray007, 2021-12-09:
@farfromrefug unfortunately, things do not work on iOS with version `3.0.32`:
```
TypeError: Cannot read property 'count' of null
```
Checking with the debugger, the exception happens in `bluetooth.ios.js` line 1411.
That is in function `_findService(UUID, peripheral)`: `peripheral.services` is `null` and so the for-loop cannot iterate.

farfromrefug, 2021-12-09:
@ray007 yes i see why. I accepted a RP i should not have ;) will fix

farfromrefug, 2021-12-10:
@ray007 3.0.33 should fix it all!

ray007, 2021-12-10:
@farfromrefug only lightly tested, but things look good for now on both platforms. Thank you.

ray007, 2022-01-20:
Upgrading from `3.0.33` to `3.1.2` the problem is back.
Only tested on android so far.

farfromrefug, 2022-01-20:
@ray007 sorry it was actually intended. `autoDiscoverAll` now defaults to false for faster device connexion. Simply add it to your `connect` options

ray007, 2022-01-20:
Hmm, I wasn't aware the `connect()` call already needs information about services which might be subsequently used, and the documentation also does not say anything about it. Is this only for non-standard services?
After looking into the lib source I added the `serviceUUIDs` parameter instead, and first testing looks good.

ray007, 2022-01-20:
Seems iOS only understands `autoDiscoverAll` and not `serviceUUIDs`, but things are now working again.
Still don't understand why `connect()` needs to know about that, but I can work with it for now.

farfromrefug, 2022-01-20:
@ray007 good catch fixed in new version! Look at the BLE docs for android. You cant access a service if it has not been discovered!

ray007, 2022-01-20:
Good try, but does not work.
Trying to use `serviceUUIDs` instead of `autoDiscoverAll` on iOS gets me an exception
```
TypeError: Cannot read property 'count' of null
```
in `_findCharacteristic(...)` in `bluetooth.ios.js` line 1461.

farfromrefug, 2022-01-20:
@ray007 try 3.1.5

ray007, 2022-01-20:
@farfromrefug Looking good, thank you.

closed 2022-01-20 by farfromrefug