nativescript-community/ble #242: peripheral doesn't contain any services (tested on android)

AvigarD611, 2022-03-15:
I have old project which is running on nativescript 7.0.0 + nodejs 12.*.* and npm 6.13.4 and everything there works perfect.

Now i tried to to upgrade the nodejs to lastest version (16.*.*) and npm is 8.1/8.2 with this version of node. After I create fresh project there are no services in peripheral object.

Steps to reproduce:
1. Create new ns project - ns create testapp
2. Enter folder - npm install
3. tns plugin add @nativescript-community/ble 
4. tns debug android

I'm using the same code from READ.me file, just removing the filters (to display all devices). After choosing mine it connects and i have console.log(peripheral) - there are no services inside. Just UUID, name and all others NO optional params.

My device is working well with my old app, which uses the same plugin. I think they've made something in core and "services" are not getting populated. Can you please check?

farfromrefug, 2022-03-15:
@AvigarD611 it is normal. Services are not discovered by default anymore as very expensive https://github.com/nativescript-community/ble/blob/master/src/bluetooth.common.ts#L362

ray007, 2022-11-25:
@farfromrefug 
> @AvigarD611 it is normal. Services are not discovered by default anymore as very expensive https://github.com/nativescript-community/ble/blob/master/src/bluetooth.common.ts#L362

The `autoDiscoverAll` property you're linking to is part of the `ConnectOptions`, not the `StartScanningOptions`.

farfromrefug, 2022-11-25:
@ray007 what s your point? you can discover services if you dont connect to a device

liamcharmer, 2023-03-24:
Sorry for being silly, how are we able to retrieve the services whether its discover or when we connect to a device?

farfromrefug, 2023-03-24:
@liamcharmer as mention here either using autoDiscoverAll option on connection or after with the discoverServices method

liamcharmer, 2023-03-24:
> @liamcharmer as mention here either using autoDiscoverAll option on connection or after with the discoverServices method

Ahhh i see sorry! The hyperlink above wasn't working and i tried my best to look at the demo but im a failure

```
 this.bluetooth.connect({
      UUID: item.UUID,
      autoDiscoverAll: true,
      onConnected: function (peripheral) {

 
        // the peripheral object now has a list of available services:
        peripheral.services.forEach(function (service) {
          console.log("service found: " + JSON.stringify(service));
        });
      },
      onDisconnected: function (peripheral) {
        console.log("Periperhal disconnected with UUID: " + peripheral.UUID);
      },
    });
    ```
    
    Many thanks as always!!!

farfromrefug, 2023-03-24:
@liamcharmer thank you and sorry too. Dont have much time for docs but should definitely improve it !