nativescript-community/ble #243: startNotifying does not respond to peripheral clicks

AustinAngular, 2022-03-27:
Hi, I am able to connect to my bluetooth remote, get back a list of services, begin startNotifying (with just one of the characteristics) but that characteristic never responds to clicking on the remote.  

this line works: console.log("startNotifying working");
but this line never responds: console.log("onNotify working");

I tried the 1812 with the 'notify: true' but that would not trigger the console.log("startNotifying working""), only the 180f/2a19 service/characteristic will trigger the console.log("startNotifying working").

the link below is the bluetooth remote I am trying to use.  It works with my phone in that from the home page the button click will increase the volume, with the camera open it will operate the shutter, but in my app it will not trigger the onNotify event callback.

https://www.amazon.com/gp/product/B07Y8M5498/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

```
ngOnInit(): void {

    console.log('ngOnInit');
    this.items = this.itemService.getItems();

    bluetooth.connect({
      UUID: '2A:07:98:10:40:05',
      autoDiscoverAll: true,
      onConnected: function (peripheral) {
        console.log("Periperhal connected with UUID: " + peripheral.UUID);

        peripheral.services.forEach(function(service) {
          console.log("service found: " + JSON.stringify(service));
        });

        bluetooth
          .startNotifying({
            peripheralUUID: '2A:07:98:10:40:05',
            serviceUUID: "180f",
            characteristicUUID: "2a19",
            onNotify: () => {
              console.log("onNotify working");
            },
          })
          .then(function () {
            console.log("startNotifying working");
          });

      },
      onDisconnected: function (peripheral) {
        console.log("Periperhal disconnected with UUID: " + peripheral.UUID);
      },
    });

  }
```

output from the periferal.services.forEach

```
service found: {"UUID":"1801","characteristics":[{"serviceUUID":"1801","UUID":"2a05","name":"2a05","properties":{"read":false,"write":false,"writeWithoutResponse":false,"notify":false,"indicate":true,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[{"UUID":"2902","value":null,"permissions":null}],"permissions":null}]}
JS: service found: {"UUID":"1800","characteristics":[{"serviceUUID":"1800","UUID":"2a00","name":"2a00","properties":{"read":true,"write":false,"writeWithoutResponse":false,"notify":false,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[],"permissions":null},{"serviceUUID":"1800","UUID":"2a01","name":"2a01","properties":{"read":true,"write":false,"writeWithoutResponse":false,"notify":false,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[],"permissions":null},{"serviceUUID":"1800","UUID":"2a04","name":"2a04","properties":{"read":true,"write":false,"writeWithoutResponse":false,"notify":false,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[],"permissions":null}]}
JS: service found: {"UUID":"180a","characteristics":[{"serviceUUID":"180a","UUID":"2a50","name":"2a50","properties":{"read":true,"write":false,"writeWithoutResponse":false,"notify":false,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[],"permissions":null}]}
JS: service found: {"UUID":"1812","characteristics":[{"serviceUUID":"1812","UUID":"2a4e","name":"2a4e","properties":{"read":true,"write":false,"writeWithoutResponse":true,"notify":false,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[],"permissions":null},{"serviceUUID":"1812","UUID":"2a4d","name":"2a4d","properties":{"read":true,"write":false,"writeWithoutResponse":false,"notify":true,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[{"UUID":"2902","value":null,"permissions":null},{"UUID":"2908","value":null,"permissions":null}],"permissions":null},{"serviceUUID":"1812","UUID":"2a4d","name":"2a4d","properties":{"read":true,"write":false,"writeWithoutResponse":false,"notify":true,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[{"UUID":"2902","value":null,"permissions":null},{"UUID":"2908","value":null,"permissions":null}],"per...
JS: service found: {"UUID":"180f","characteristics":[{"serviceUUID":"180f","UUID":"2a19","name":"2a19","properties":{"read":true,"write":false,"writeWithoutResponse":false,"notify":true,"indicate":false,"broadcast":false,"authenticatedSignedWrites":false,"extendedProperties":false},"descriptors":[{"UUID":"2902","value":null,"permissions":null}],"permissions":null}]}
```

mst-marcernst, 2022-05-25:
Does your device support notifications on the battery service? According to the specification, notify is optional. Try read the characteristic instead?

farfromrefug, 2022-11-27:
@AustinAngular i will close this issue sa it is not an issue with the plugin but more communicating with a device.
You can continue that discussion on N discord which is just for that

closed 2022-11-27 by farfromrefug