nativescript-community/ble #246: startScanning fails on Android 12

mst-marcernst, 2022-05-25:
Hello!

Thank you for making this project available! I've been successfully working with this on projects compiling against Android versions below 12, but am now receiving the below error when setting up a new nativescript project using the 3.1.8 plugin version on Android 12 (compileSdkVersion 31).

I'm ensuring location permissions have been granted via hasLocationPermission & requestLocationPermission, but am receiving the below error when calling startScanning.

java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@472b4357: GattService registerScanner, startScanning, undefined, {"seconds":10}

I've added the BLUETOOTH_SCAN permission to the manifest, to no avail. Any idea?

Thanks! ME

farfromrefug, 2022-05-25:
@mst-marcernst you need to read android docs https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

mst-marcernst, 2022-05-25:
Thank you for your quick response. But permissions are handled within the BLE plugin itself, it was only required to add e.g. ACCESS_COARSE_LOCATION and/or ACCESS_FINE_LOCATION to the android manifest. I've tried adding android.permission.BLUETOOTH_SCAN to the manifest, to no avail.

mst-marcernst, 2022-05-25:
That makes very little sense, doesn't it? Your argument that "there are too many different cases" actually creates a very strong case for handling this within the plugin itself. That's why the plugin offers methods such as 'hasLocationPermission' and 'requestLocationPermission'. BLUETOOTH_SCAN is a vital requirement for the startScanning method to work under Android 12, and thus should be handled by the plugin, too.

farfromrefug, 2022-05-25:
@mst-marcernst it is the total contrary. If i handle this in the plugin then it will force one way of doing it while apps might want to do it differently.
Your argument "it is vital so the plugin must do it" is not valid. Anything in your app is vital so should the plugin handle it all?
This is not  made to handle permissions. As i said it is still here for compatibility reasons. There are other plugins to handle permissions and i wont duplicate code nor will i add logic which i think should not be here. Sorry about that.

closed 2022-05-25 by farfromrefug