nativescript-community/ble #247: Support Android 12

JakeAi, 2022-05-25:
I can't get my app to work on Android 12. I have followed the questions online (unrelated to nativescript) but I get this error everytime I try and scan.

```
Error: java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for AttributionSource { uid = 10642, packageName = com.my.app, attributionTag = null, token = android.os.BinderProxy@1ca5d6c, next = null }: GattService registerScanner
```

farfromrefug, 2022-05-25:
@JakeAi As explained here https://developer.android.com/guide/topics/connectivity/bluetooth/permissions you need to ask permission at runtime. The plugin wont handle the permission check. You can use @nativescript-community/perms

JakeAi, 2022-05-25:
@farfromrefug is it as easy as ?
```TS
    await request('bluetooth');
    await request('bluetoothConnect');
    await request('bluetoothScan');
```

farfromrefug, 2022-05-25:
@JakeAi yes it is :D Dont forget to add the perms in your manifest

JakeAi, 2022-05-25:
@farfromrefug might be a good idea to add it to the readme?

closed 2022-05-25 by JakeAi

farfromrefug, 2022-05-25:
@JakeAi Yes you can create a PR it will be welcome. But if you do so please add a link to the android doc. I understand the need for this in the readme but i would really wish people first look at android doc to understand what they need to do

JakeAi, 2022-05-25:
> @JakeAi Yes you can create a PR it will be welcome. But if you do so please add a link to the android doc. I understand the need for this in the readme but i would really wish people first look at android doc to understand what they need to do

Well, I did read the doc. I had the manifest updated a while ago. The problem was manually requesting permissions at runtime.
There are a lot of plugins. That's all. With the plugin being called "perms" makes it a little hard to find as well lol.
[https://plugins.nativescript.rocks/search/permissions](https://plugins.nativescript.rocks/search/permissions)

farfromrefug, 2022-05-25:
@JakeAi this is was not a critic for you or anyone else. I am just saying the first step should always be to read android/ios docs. 
About the perms plugin. Indeed you are right it would be a good idea to mention it! it is used internally in this plugin.
There are 2 different plugins indeed, and i would personally avoid `@master.technology/permissions`. It is not open source, not free and not implemented in a true cross platform way