nativescript-community/ble #248: onDisconnected not called after killing the gatt server

jnorkus, 2022-06-02:
I am only starting with BLE and I might be missing something here. I am looking for a way to consistently detect disconnects even if the connection was not closed gracefully.

As per title, I am not getting any kind of indication when killing the server. The callback is not getting called, `isConnected` always returns `true`.

farfromrefug, 2022-06-02:
@jnorkus i dont see how you expect it work if you kill the GATT server. why would you do that ? killing the app and killing the GATT server are very different. killing the app might happen but you don't care cause your app is killed... killing GATT should not happen and would mean (not 100% sure) needing to restart the app for the ble plugin state machine to restart

jnorkus, 2022-06-03:
@farfromrefug By killing the GATT server I'm simulating a loss of connection. In real life that could be e.g. out of range.

farfromrefug, 2022-06-03:
@jnorkus no you are not it is very different. loss of connection is handled by the GATT server. without the GATT server you can't be notified. out or range === manually turning off your device without it sending any notif.
be aware connection loss notification in those cases can take sometimes up to a minute or so

jnorkus, 2022-06-03:
Thanks @farfromrefug I think I understand the issue now. The BLE connection is not terminated in case the GATT server is shut down, that effectively means it's still connected. Turning off BT on the server peripheral works fine and the `onDisconnect` callback is called.

closed 2022-06-03 by jnorkus