nativescript-community/ble #249: TypeError: Cannot read property 'array' of undefined

JakeAi, 2022-06-16:
I upgraded ble from 3.1.7 to 3.1.8, and my production builds produce this exception
```
TypeError: Cannot read property 'array' of undefined
  at pointsFromBuffer(vendor.js:2:19971)
  at arrayToNativeArray(vendor.js:2:20093)
  at writeValueToChar(vendor.js:2:50841)
  at ? (vendor.js:2:52431)
  at attachSubDelegate(vendor.js:2:64076)
  at ? (vendor.js:2:52009)
  at new ZoneAwarePromise(vendor.js:2:1814383)
  at ? (vendor.js:2:51466)
  at ? (vendor.js:2:47209)
  at invoke(vendor.js:2:1801295)
```
```
BluetoothService Write
{
    characteristicUUID: 777FC661-82E2-4BC2-B189-C816EEF342D2, 
    peripheralUUID: 00:A0:XX:XX:XX:XX, 
    serviceUUID: 8BCABDAA-FACB-4E37-B92A-8590855EFEB2, 
    value: {
        name_value_pairs: {
            data: {
                values: [187,132,15,186,132,6,251,151,140,95,201,33,171,33,151,225,0,0,0,0]
            }, 
        type: Buffer
        }
    }
}
```

JakeAi, 2022-06-16:
> @JakeAi What is that value your are trying to send?

Its a device unlock key that is a Uint8Array and then a Buffer

JakeAi, 2022-06-16:
I don't understand why it only throws on production builds and not when I'm debugging.

JakeAi, 2022-06-20:
> @JakeAi sorry without seeing your code i cant help. you need to deubg what s gooing one . You can run your app with `--release` so run in production.
> 
> I will close that isssue as it is mostly surely not releated to this plugin as it works in non production builds

    let serialTemp: string = '1010-000100';
    const unlockKey = [0x8a, 0xb4, 0x3d, 0x8a, 0xa9, 0xa9, 0xa9, 0xa9, 0xbe, 0x6f, 0xf9, 0x10, 0x9b, 0x13, 0xa7, 0xcc];

    while (serialTemp.length < 16) {
      serialTemp += serialTemp;
    }

    const keyCodeBuffer = new Uint8Array(16);
    serialTemp = serialTemp.trim().substring(0, 16);

    for (let i = 0; i < 16; i++) {
      keyCodeBuffer[i] = ((serialTemp.charCodeAt(i) & 0xff) >>> 0) ^ (unlockKey[i] >>> 0);
    }

    console.log(pointsFromBuffer(keyCodeBuffer));
    console.log(pointsFromBuffer(new Buffer(keyCodeBuffer)));

JakeAi, 2022-06-21:
@farfromrefug I find it hard to believe it's not the new arraybuffers plugin.

ns run --release
```TS
{runtimeVersion:"0.0.0.0", supportsDirectArrayBuffers: false}
```

ns build --release
```TS
{runtimeVersion:"0.0.0.0", supportsDirectArrayBuffers: false}
```

ns debug
```TS
{runtimeVersion:"8.2.3", supportsDirectArrayBuffers: true}
```

sebj54, 2022-07-14:
@JakeAi for the record, it will be fixed in the next @nativescript/android version. See https://github.com/NativeScript/android-runtime/issues/1715

msn444, 2022-07-27:
@sebj54 Is this still forthcoming? I'm having the same issue on `@nativescript/android 8.3.0`.

sebj54, 2022-07-28:
I just tested with a sample app built in release mode and it has the correct runtime version. Did you run `ns clean` before the build command?