nativescript-community/ble #251: Android app crashes while updating

jnorkus, 2022-11-14:
We have been observing multiple crashes in production during a new version rollout in Google Play Store. It happens while the app is running and connected via BLE. The app crashes with the following error messages:
```
Fatal Exception: com.tns.NativeScriptException: Calling js method onCharacteristicChanged failed
TypeError: Cannot read property 'connections' of null
```
```
Fatal Exception: com.tns.NativeScriptException: Calling js method onCharacteristicChanged failed
RangeError: index out of range: 7 + 1 > 7
```

I have previously reported [this issue](https://github.com/nativescript-community/ble/issues/250), which is very similar.

farfromrefug, 2022-11-27:
@jnorkus i fixed in 3.1.1 however i want to mention that those errors means that you are not disconnecting from your device on app close which seems dangerous.

closed 2022-11-27 by farfromrefug

jnorkus, 2022-11-28:
Thanks, you're right, I'm not disconnecting as the disconnection just happens automatically while closing the app. I'm not exactly sure how the update process works on Android but looks like this disconnection does not happen in the same way.

Do you have any suggestions on the implementation? Is `Application.exitEvent` a good place for such cleanup?

farfromrefug, 2022-11-28:
@jnorkus yes i would do it this way.
In my apps i close my "service" (not angular service, more like android service) like this https://github.com/Akylas/jule_verne/blob/ae5c23e1d48326241d630543fe5d0116d7948144/app/services/BgService.common.ts#L81
which then stops all my handlers (name i use might not be perfect)like gps, db, ble... 
And for the ble handler i there disconnect any ble connection https://github.com/Akylas/jule_verne/blob/ae5c23e1d48326241d630543fe5d0116d7948144/app/handlers/BluetoothHandler.ts#L242