nativescript-community/ble #253: startScanning doing nothing

erik777, 2022-11-21:
I get nothing back via onDiscovered, nor the then or error result of startScanning...

```
          console.log("startScanning - calling");
          this.btInstance
              .startScanning({
                  seconds: 4,
                  onDiscovered: (perip: Peripheral) => {
                    console.log(`onDiscovered()`);
                  }
              })
              .then(() => {
	                console.log(`startScanning - then`);
	            	  this.isLoading = false;
	              },
                (err: any) => {
                  console.log(`startScanning - err`);
                }
              );
          console.log("startScanning - called");
```

Note that I do get a then called for btInstance.enable() with true when on mobile, false when on the emulator.  The then on btInstance.stopScanning() is also successfully called.  I receive a console.log for that.  

But, for startScanning, the only logging I'm getting is before and after it is called.

```
  JS: startScanning - calling
  JS: startScanning - called
```

Nothing happens after 4 seconds, either.  I started with 50 seconds and reduced over time.

farfromrefug, 2022-11-21:
@erik777 did you request the correct permissions ? Android 11+ requires new permissions.

erik777, 2022-11-21:
> @erik777 did you request the correct permissions ? Android 11+ requires new permissions.

I thought the prompt for location access was enough, which i grant.  Do I need to add code to explicitly request permissions?  This is in debug mode.  

Shouldn't the call to startScanning result in an error if permissions are not granted?  The promise is not rejected.

farfromrefug, 2022-11-21:
The demo is quite old and android is moving really fast. It is best or you to look at android docs to understand things and what perms are needed. Then you can use @nativescript-community/perms to ask for the permission

erik777, 2022-11-22:
Thanks for your help.  I installed and got perms working.  When I check perms, it says it is authorized for "bluetooth" and "bluetoothScan", and "undetermined" for location.  Then, `requestPermission(['location', 'bluetooth', 'bluetoothScan']` says "authorized" for course location, bluetooth scan and bluetooth advertise.  When I drill into App settings permissions, it has the same permissions as my working Cordova app, including "Location" and "Nearby Devices".  It's looking less like an issue with perms.

farfromrefug, 2022-11-22:
@erik777 did not know about the neverForLocation breaking scanning. Always use both ;)
Thanks for explaining here might help others !

erik777, 2022-11-23:
Yeah, scanning is working now.  Thanks for your help!  I'll worry about fine-tuning it later (perhaps with filters) to try to reduce permission requirements.  

A little off-topic, so if you want me to create another issue to ask, I can...  

The ultimate goal is to provide a P2P service with devices automatically discovering each other.  I'm a bit new to bluetooth, but doesn't that mean I'll need the ADVERTISE ability?  I don't see it exposed in the ble library.  Or, is there another way to accomplish this?  

If it is not in ns-ble, I'm opening to learning how to expose it and creating a PR, though that could take some time.

farfromrefug, 2022-11-27:
@erik777 can we close this?

erik777, 2022-12-22:
Yeah.  I have a new issue with services, but I'll open a new ticket for it.

closed 2022-12-22 by farfromrefug