nativescript-community/ble #256: bluetooth.connect with autoDiscoverAll hangs on iOS for BT Assigned Numbers

two-bridges, 2022-12-04:
my test project:
```json
{
        "@nativescript/core": "^8.4.1",
        "@nativescript-community/ble": "^3.1.11",
}
```

I'm investigating an issue where bluetooth.connect() fails to resolvee the promise when a serviceUUID falls under the Bluetooth Assigned Numbers list: 
https://www.bluetooth.com/specifications/assigned-numbers/

I have tracked the problem down to:
https://github.com/nativescript-community/ble/blob/master/src/bluetooth.ios.ts#L1643-L1644 

```typescript
                    const sUUID = CBUUIDToString(service.UUID);
                    if (UUID === pUUID && sUUID === args.serviceUUID) {
                    }
```

the issue is the following comparison fails:
```typescript
# "0000181a-0000-1000-8000-00805f9b34fb" == "181a"
sUUID === args.serviceUUID
```

args.serviceUUID is being be modified by the decorator `prepareArgs()` to a "Short UUID", if the service's UUID matches a Bluetooth Assigned Number of the form.  

howeveer, sUUID returned by `CBUUIDToString(service.UUID)` is a "Long UUID"

I'll work on a fix and submit a PR today hopefully...

two-bridges, 2022-12-04:
PR created in forked repo (I don't have access to this repo):

https://github.com/two-bridges/ble/pull/1

I'm testing it now...

two-bridges, 2022-12-04:
my local testing passed.  I'm not sure how to automate the testing.  Since, it required physical BLE device and physical iPhone to get the result... 

Anyway, could you please review the code and consider if/how to merge this into a future version 🙏 

Thanks

farfromrefug, 2022-12-04:
@two-bridges i like the fact that you now compare long and short. Seems safer. Though what s weird is that I know for sure it was working for me for battery and other BAN. Might it be in a new iOS version?
Anyway your PR looks good. Could you open it on this repo?

two-bridges, 2022-12-04:
Yeah I have been using this plugin for a while and this issue was new for me... 

There are two things that have changed on my side that could have caused this:
1. iPhone upgrade to latest 16.x
2. the arduino BLE programming is new (from a contractor). This programming should not really be using an Assigned Name actually... but here we are.  

I cannot tell if (1) or (2) caused this issue, since they have both changed since I last did a build.  

> i like the fact that you now compare long and short. Seems safer

cool cool.  I think this is the way to go too... should be backwards compatible