nativescript-community/ble #264: error_function_call

lvrnls, 2023-01-25:
Hi,
I'm trying to write data to a characteristic of a BLE device (following the doc I have for this) but I always get an error :
```
ERROR Error: Uncaught (in promise): [BluetoothError]:error_function_call, writeCharacteristic, undefined, {"peripheralUUID":"A4:C1:38:C4:28:8A","serviceUUID":"1000","characteristicUUID":"1001","value":{}}
JS: BluetoothError: error_function_call
JS:     at file: src\webpack:\myapp\node_modules\@nativescript-community\ble\index.android.js:1934:32
JS:     at Bluetooth.attachSubDelegate (file: src\webpack:\myapp\node_modules\@nativescript-community\ble\index.android.js:2388:12)
JS:     at file: src\webpack:\myapp\node_modules\@nativescript-community\ble\index.android.js:1897:17
JS:     at new ZoneAwarePromise (file: src\webpack:\myapp\node_modules\zone.js\fesm2015\zone.js:1429:0)
JS:     at file: src\webpack:\myapp\node_modules\@nativescript-community\ble\index.android.js:1874:50
JS:     at file: src\webpack:\myapp\node_modules\@nativescript-community\ble\index.android.js:1633:82
JS:     at _ZoneDelegate.invoke (file: src\webpack:\myapp\node_modules\zone.js\fesm2015\zone.js:372:0)
JS:     at Object.onInvoke (file: src\webpack:\myapp\node_modules\@angular\core\fesm2015\core.mjs:25548:0)
JS:     at _ZoneDelegate.invoke (file: src\webpack:\myapp\node_modules\zone.js\fesm2015\zone.js:371:0)
JS:     at Zone.run (file:///data/data/be....
```
  
Here is what I try to do in the onConnected method :
```
let data = new Uint8Array([0x07, 0x04, 0x03, 0x00, 0x1e]);
self.bluetooth.writeWithoutResponse({
    peripheralUUID: peripheral2.UUID,
    serviceUUID: "00001000-0000-1000-8000-00805f9b34fb",
    characteristicUUID: "00001001-0000-1000-8000-00805f9b34fb",
    value: data.buffer
}).then(() => {
...
});
```

I'm having a similar issue with the read function 
`JS: read error: [BluetoothError]:error_function_call, readCharacteristic, undefined, {"peripheralUUID":"A4:C1:38:FD:D2:38","serviceUUID":"1000","characteristicUUID":"1002"}`

Here is my code for that
```
self.bluetooth.read({
  peripheralUUID: peripheral2.UUID,
  serviceUUID: '00001000-0000-1000-8000-00805f9b34fb',
  characteristicUUID: '00001002-0000-1000-8000-00805f9b34fb',
}).then(function(result) {
...
});
```

I tried to read/write to those characteristics with a Java application and it's working so I guess this is a plugin issue ? Can you help me with that if it is ? 
I use Nativescript with Angular and the phone that connects to the device is on Android 12.

Thanks in advance !

farfromrefug, 2023-01-27:
@lvrnls did you discover the service/char before?
THis is a requirement on android (see android bluetooth doc)

lvrnls, 2023-01-27:
@farfromrefug Yes I think that part should be ok, I used the serviceUUIDs option in the connect method like this
```
self.bluetooth.connect({
            UUID: self.periphs[0].UUID,
            serviceUUIDs: ['00001000-0000-1000-8000-00805']
```

Is that what you meant ?

farfromrefug, 2023-01-27:
@lvrnls no it is not done automatically. Look a the typings for connect options. THere is one to discover all services. Though it will be slower. The best way is to call discoverServices once connected and discover only the services you need

lvrnls, 2023-01-27:
Ok thanks, I did
```
self.bluetooth.discoverServices({
    peripheralUUID: peripheral2.UUID,
    serviceUUIDs: ["00001000-0000-1000-8000-00805f9b34fb"]
}).then((serv) => {
...
});
```
before trying to write or read but I still got the issue `JS: read error: [BluetoothError]:error_function_call, readCharacteristic, undefined, {"peripheralUUID":"A4:C1:38:E9:D2:5C","serviceUUID":"1000","characteristicUUID":"1002"}`

farfromrefug, 2023-01-27:
@lvrnls look at the error maybe try to pass `1000` as `serviceUUIDs`. appart from that i dont really know

lvrnls, 2023-02-08:
@farfromrefug Thank you for your help anyway, I've contacted the manufacturer to see if something comes from him.