nativescript-community/ble #267: (iOS) startScanning() not working

raphcibo1, 2023-02-27:
Hi,

I'm facing an issue that I can't explain when trying to get info on a BLE device. StartScanning() seems to not doing anything only on iOS. On Android, everything works as expected.

Here's my method and on iOS, nothing comes from onDiscovered :

```
bleRoutine(){
        try{
            let self = this;
            this.bluetooth.startScanning({
                //filters: [{serviceUUID:'00001000-0000-1000-8000-00805f9b34fb'}],
                seconds: 3,
                avoidDuplicates: true,
                onDiscovered: function (peripheral) {
                    if(peripheral.name == "T201"){
                        let dataView = new DataView(peripheral.advertismentData.manufacturerData);
                    }
                }
            }).then(function() {
                //console.log("scanning complete");
                //if(!self.bleStop) self.bleRoutine();
                // self.connectToDevice();
                //self.bleConnections();
            }, function (err) {
                //console.log("error while scanning: " + err);
            });
        }catch(e){
            console.log("ble catch");
            console.log(e);
        }
    }
```

Here's what I added in my Info.plist :

 ```
<key>NSBluetoothAlwaysUsageDescription</key>
<string>Our app uses bluetooth to find, connect and transfer data between different devices</string>
<key>NSBluetoothPeripheralUsageDescription</key>  
<string> Our app uses bluetooth to find, connect and transfer data between different devices</string>
<key>NSLocationWhenInUseUsageDescription</key>  
<string> Our app uses location</string>
```

I also checked `Uses Bluetooth LE accessory` in Background Modes in Xcode.

I also have a bluetooth message when I start the app as expected. I am in native script version 8.1.4 and `"@nativescript-community/ble": "^3.1.15"`

Am I missing something ? (maybe something obvious that doesn't ring a bell in my head)

farfromrefug, 2023-02-27:
@raphcibo1 did you ask for location permission at runtime? it is needed to be able to scan
also did you add a simple console.log in `onDiscovered` to see if it sees something?

NathanWalker, 2023-05-07:
Another things to check here is the name adding spaces from the device, for example this conditional:
```
if(peripheral.name == "T201"){
```
may be causing you to think no devices are received - it's common the bluetooth device names may have leading or trailing spaces, eg: `"T201      "`