nativescript-community/ble #268: TypeError: Cannot read properties of undefined (reading 'toLowerCase')

JakeAi, 2023-04-03:
After an xcode update and the cocopod fiasco, I am now unable to use this plugin.

The error is coming from function CBUUIDToString when trying to scan.
```
  TypeError: Cannot read properties of undefined (reading 'toLowerCase')
  at CBUUIDToString (file:///app/vendor.js:1187:26)
  at get serviceUUIDs [as serviceUUIDs] (file:///app/vendor.js:1112:21)
  at file:///app/bundle.js:842:67
  at Array.map (<anonymous>)
```

farfromrefug, 2023-04-03:
@JakeAi Did you by any change upgrade the ios runtime to 8.5.x ?
If so could you confirm it works with 8.4.x ?

JakeAi, 2023-04-03:
> @JakeAi Did you by any change upgrade the ios runtime to 8.5.x ? If so could you confirm it works with 8.4.x ?

I am using 8.5.0. However, I am updating my device to 16.4.

JakeAi, 2023-04-03:
> @JakeAi we are talking @nativescript/ios right? If so please try to use 8.4.x, clean and let me know if it works. I might now what s happening

While I was messing around with the versions, I noticed there was a 8.5.1 for iOS. Would this fix the issue I was having?

https://github.com/NativeScript/ios/commit/b6894346b273b289bfda713a8f0c7055911c945a

farfromrefug, 2023-04-03:
@JakeAi yes that could really be it!

JakeAi, 2023-04-03:
> @JakeAi yes that could really be it!

Unreal, that indeed has fixed my issue lol. It wasn't released yet when I updated everything to 8.5.0 lol.

closed 2023-04-03 by JakeAi