nativescript-community/ble #271: App crash on write

EikiProg, 2023-06-16:
Hello,

@nativescript-community/ble": "3.1.15"
@nativescript/core": "8.5.3"
compileSdkVersion 34
buildToolsVersion "34.0.0"
minSdkVersion 17
targetSdkVersion 34

After i do bluetooth connect then startNotifying, if i try to do write or writeWithoutResponse like this

```js
bluetooth.writeWithoutResponse({
  peripheralUUID: '04:91:62:9A:16:9B',
  serviceUUID: '49535343-fe7d-4ae5-8fa9-9fafd205e455',
  characteristicUUID: '49535343-1E4D-4BD9-BA61-23C647249616',
  value: ['ADR11BC2BEF2E9BF7530']
})
```

the app crash with a JNI error

```bash
06-16 18:21:49.779 E/.softconnectdev(24281): JNI ERROR (app bug): attempt to pass an instance of int[] as argument 2 to boolean com.nativescript.bluetooth.Bluetooth.writeCharValue(android.bluetooth.BluetoothGattCharacteristic, byte[])
06-16 18:21:49.779 F/.softconnectdev(24281): java_vm_ext.cc:594] JNI DETECTED ERROR IN APPLICATION: bad arguments passed to boolean com.nativescript.bluetooth.Bluetooth.writeCharValue(android.bluetooth.BluetoothGattCharacteristic, byte[]) (see above for details)
06-16 18:21:49.779 F/.softconnectdev(24281): java_vm_ext.cc:594]     from java.lang.Object com.tns.Runtime.callJSMethodNative(int, int, java.lang.String, int, boolean, java.lang.Object[])
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] Runtime aborting --- recursively, so no thread-specific detail!
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #00 pc 00000000006b1320  /apex/com.android.art/lib64/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+128) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #01 pc 00000000006ff92c  /apex/com.android.art/lib64/libart.so (art::AbortState::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char> >&) const+88) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #02 pc 00000000006fa114  /apex/com.android.art/lib64/libart.so (art::Runtime::Abort(char const*)+964) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #03 pc 0000000000016ea8  /apex/com.android.art/lib64/libbase.so (android::base::SetAborter(std::__1::function<void (char const*)>&&)::$_3::__invoke(char const*)+80) (BuildId: 420d56eac27a210c92900f3ddb760c86)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #04 pc 0000000000016450  /apex/com.android.art/lib64/libbase.so (android::base::LogMessage::~LogMessage()+352) (BuildId: 420d56eac27a210c92900f3ddb760c86)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #05 pc 00000000004664e0  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+2116) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #06 pc 000000000034cbec  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbortV(char const*, char const*, std::__va_list)+108) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #07 pc 000000000056896c  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbortF(char const*, char const*, ...)+112) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #08 pc 0000000000465838  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckMethodArguments(art::JavaVMExt*, art::ArtMethod*, unsigned int*) (.__uniq.245181933781456475607640333933569312899.llvm.14826870601531466521)+1460) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #09 pc 00000000003bcb04  /apex/com.android.art/lib64/libart.so (art::JValue art::InvokeWithJValues<art::ArtMethod*>(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, art::ArtMethod*, jvalue const*)+1124) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #10 pc 00000000006827f8  /apex/com.android.art/lib64/libart.so (art::JNI<true>::CallStaticBooleanMethodA(_JNIEnv*, _jclass*, _jmethodID*, jvalue const*)+164) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #11 pc 000000000034ab20  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::CallMethodA(char const*, _JNIEnv*, _jobject*, _jclass*, _jmethodID*, jvalue const*, art::Primitive::Type, art::InvokeType) (.__uniq.99033978352804627313491551960229047428)+1584) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #12 pc 0000000000343d6c  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::CallStaticBooleanMethodA(_JNIEnv*, _jclass*, _jmethodID*, jvalue const*) (.__uniq.99033978352804627313491551960229047428.llvm.16720480732663731947)+44) (BuildId: 289d75599f6112d5757113220599e90b)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #13 pc 000000000060e530  /data/app/~~8WutuywLd8k5Pj5cBtjMkA==/org.nativescript.softconnectdev-0EYdCO_p47B3cTlktW27lg==/lib/arm64/libNativeScript.so (tns::JEnv::CallStaticBooleanMethodA(_jclass*, _jmethodID*, jvalue*)+28) (BuildId: 9f9c59847af6c0b0b0fbeb9aa95cb67b3e7e1a07)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #14 pc 00000000005ff380  /data/app/~~8WutuywLd8k5Pj5cBtjMkA==/org.nativescript.softconnectdev-0EYdCO_p47B3cTlktW27lg==/lib/arm64/libNativeScript.so (tns::CallbackHandlers::CallJavaMethod(v8::Local<v8::Object> const&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&, tns::MetadataEntry*, bool, bool, bool, v8::FunctionCallbackInfo<v8::Value> const&)+2480) (BuildId: 9f9c59847af6c0b0b0fbeb9aa95cb67b3e7e1a07)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #15 pc 0000000000626900  /data/app/~~8WutuywLd8k5Pj5cBtjMkA==/org.nativescript.softconnectdev-0EYdCO_p47B3cTlktW27lg==/lib/arm64/libNativeScript.so (tns::MetadataNode::MethodCallback(v8::FunctionCallbackInfo<v8::Value> const&)+600) (BuildId: 9f9c59847af6c0b0b0fbeb9aa95cb67b3e7e1a07)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #16 pc 0000000000df9a8c  /data/app/~~8WutuywLd8k5Pj5cBtjMkA==/org.nativescript.softconnectdev-0EYdCO_p47B3cTlktW27lg==/lib/arm64/libNativeScript.so (v8::internal::FunctionCallbackArguments::Call(v8::internal::CallHandlerInfo)+316) (BuildId: 9f9c59847af6c0b0b0fbeb9aa95cb67b3e7e1a07)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #17 pc 0000000000df91a8  /data/app/~~8WutuywLd8k5Pj5cBtjMkA==/org.nativescript.softconnectdev-0EYdCO_p47B3cTlktW27lg==/lib/arm64/libNativeScript.so (???) (BuildId: 9f9c59847af6c0b0b0fbeb9aa95cb67b3e7e1a07)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #18 pc 0000000000df87d0  /data/app/~~8WutuywLd8k5Pj5cBtjMkA==/org.nativescript.softconnectdev-0EYdCO_p47B3cTlktW27lg==/lib/arm64/libNativeScript.so (???) (BuildId: 9f9c59847af6c0b0b0fbeb9aa95cb67b3e7e1a07)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] #19 pc 0000000000ccfa48  /data/app/~~8WutuywLd8k5Pj5cBtjMkA==/org.nativescript.softconnectdev-0EYdCO_p47B3cTlktW27lg==/lib/arm64/libNativeScript.so (???) (BuildId: 9f9c59847af6c0b0b0fbeb9aa95cb67b3e7e1a07)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:681] 
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:689] JNI DETECTED ERROR IN APPLICATION: bad arguments passed to boolean com.nativescript.bluetooth.Bluetooth.writeCharValue(android.bluetooth.BluetoothGattCharacteristic, byte[]) (see above for details)
06-16 18:21:49.799 F/.softconnectdev(24281): runtime.cc:689]     from java.lang.Object com.tns.Runtime.callJSMethodNative(int, int, java.lang.String, int, boolean, java.lang.Object[])
```
The characteristics are like this

```json
{
   "serviceUUID":"49535343-fe7d-4ae5-8fa9-9fafd205e455",
   "UUID":"49535343-1e4d-4bd9-ba61-23c647249616",
   "name":"49535343-1e4d-4bd9-ba61-23c647249616",
   "properties":{
      "read":false,
      "write":true,
      "writeWithoutResponse":true,
      "notify":true,
      "indicate":true,
      "broadcast":false,
      "authenticatedSignedWrites":false,
      "extendedProperties":false
   },
   "descriptors":[
      {
[... 11 lines cut ...]
[... cut ...]
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.REQUEST_ENABLE_LOCATION"/>

<uses-feature android:name="android.hardware.location.network" android:required="true" />
<uses-feature android:name="android.hardware.touchscreen" android:required="true" />
<uses-feature android:name="android.hardware.bluetooth" android:required="true" />
<uses-feature android:name="android.hardware.location" android:required="true" />
```
The phone i test with is a Pixel 6a with android 13

farfromrefug, 2023-06-16:
@EikiProg your value parameter is wrong. You need to pass an arraybuffer

EikiProg, 2023-07-06:
Thank you, with the use of [TextEncoder](https://developer.mozilla.org/en-US/docs/Web/API/TextEncoder), i could solve my problem.

closed 2023-07-06 by EikiProg