nativescript-community/ble #272: scanning does nothing and does not stop after specified time

elsni, 2023-06-20:
To get started I want to scan for devices. Unfortunately the scan seems to do nothing. I does not stop after 4 seconds.
No devices are being found, no error occurs, just nothing.

Stopping via code produces no error, but seems to do nothing else either.

Tested on Android: Samsung Galaxy A13

main-page.xml:
```
<Page xmlns="http://schemas.nativescript.org/tns.xsd" navigatingTo="navigatingTo" loaded="loaded">
    <ActionBar title="X-am 8000 ble connect" icon="" />
    <ScrollView>
        <StackLayout class="p-20" orientation="vertical">
            <Label text="Blestatus" class="h1 text-center" />
            <Label text="{{ blestatus }}" class="h2 text-center" textWrap="true" />
            <Button text="{{ blebuttontext }}" tap="{{ onTap }}" class="-primary" id="scanbutton" />
            <ListView items="{{ deviceList }}" height="1000">
                <ListView.itemTemplate>
                    <StackLayout>
                        <Label text="{{ name }}" textWrap="false" horizontalAlignment="left"/>
                        <Label text="{{ mac }}" textWrap="true" horizontalAlignment="left"/>
                    </StackLayout>
                </ListView.itemTemplate>
            </ListView>
        </StackLayout>
    </ScrollView>
</Page>

```

main-page.ts:
```
import { EventData, Page } from '@nativescript/core'
import { BleModel } from './main-view-model'

export function navigatingTo(args: EventData) {
  console.log("page navigated to");

}

export function loaded(args: EventData) {
  console.log("page loaded");
  const page = <Page>args.object;
  const model = new BleModel();
  page.bindingContext = model;
}
```

main-view-model.ts

```
import { Button, EventData, Observable, ObservableArray } from '@nativescript/core'
import { Bluetooth, Peripheral, getBluetoothInstance } from '@nativescript-community/ble';

export type device = {
  name: string,
  mac: string
}

export class BleModel extends Observable {
  private _bleenabled: boolean;
  private _permissionsgranted: boolean;
  private _message: string;
  private _devices: ObservableArray<device>;
  private _bluetooth = getBluetoothInstance();
  private _scanning = false;
  private _blebttext="start scan";

  constructor() {
    super();
    this._devices = new ObservableArray<device>;
    this._bluetooth = new Bluetooth();
    this._bluetooth.hasLocationPermission().then(
      function (granted) {
        if (!granted) {
          this._bluetooth.requestLocationPermission().then(function (granted) {
            this._permisionsgranted = true;
          });
        }
      }
    );

    this._bluetooth.isBluetoothEnabled().then(
      function (enabled) {
        this._bleenabled = enabled;
      }
    );
   console.log("constructor done");
  }
  get deviceList(): ObservableArray<device> {
    return this._devices;
  }

  get blebuttontext():string {
    return this._blebttext;
  }
  set blebuttontext(txt:string) {
    this._blebttext=txt;
    this.notifyPropertyChange('blebuttontext',txt);
  }
  get bleenabled(): boolean {
    return this._bleenabled;
  }
  get permisionsgranted(): boolean {
    return this._bleenabled;
  }
  set permisionsgranted(value: boolean) {
    this._permissionsgranted = value;
  }

  set bleenabled(value: boolean) {
    this._bleenabled = value;
    this.notifyPropertyChange('blestatus', value ? 'enabled' : 'disabled');
  }

  onTap(args: EventData) {
    if (this._scanning) {
      this._bluetooth.stopScanning().then(function() {
        console.log("scanning stopped");
        //this.notifyPropertyChange('blebuttontext', 'start scan');
        this.blebuttontext ="start scan";
        this._scanning = false;
      });
    }
    else {
      console.log("start scanning...");
      this.blebuttontext ="stop scan";
      //this.notifyPropertyChange('blebuttontext', 'stop scan');
      this._scanning = true;
      this._bluetooth.startScanning({
        seconds: 4,
        skipPermissionCheck: false,
        onDiscovered: function (peripheral) {
          console.log("found device " + peripheral.localName + " mac " + peripheral.UUID + " name " + peripheral.name);
          //this._devices.push({ name: peripheral.localName, mac: peripheral.UUID });
        }
      }).then(function () {
        console.log("scanning complete");
      }, function (err) {
        console.log("error while scanning: " + err);
      });
    }
  }
}
```

farfromrefug, 2023-06-20:
@elsni did you ask of all needed runtime permissions? You should look at android docs for ble

elsni, 2023-06-21:
Yes, bluetooth and bluetooth-admin. But even if the permission were missing, "nothing" should not happen.

elsni, 2023-06-21:
Found bug in the code, and updated it above. There indeed is a permission issue in addition.
`Need android.permission.BLUETOOTH_SCAN permission for AttributionSource `
but I already added 
`<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />`
to manifest.

farfromrefug, 2023-06-21:
@elsni you need to ask for the permission at runtime! you can use @nativescript-community/perms

elsni, 2023-06-22:
ah thank you - sorry I'm new to this.
`requestPermission({bluetoothScan:{}, bluetoothConnect:{}})`
does the trick

closed 2023-06-22 by elsni