nativescript-community/ble #276: Scan throws error on android 13+

CatchABus, 2023-12-11:
It seems that android 13+ devices need an additional permission for BLE scan.
Error log:
```sh
Error: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{e7d2f4d 5142:com.fuelics.mobile.alpha/u0a159} (pid=5142, uid=10159) requires android.permission.BLUETOOTH_CONNECT undefined
```

Google claimed that this permission was useful for devices that were already paired in API 31 https://developer.android.com/develop/connectivity/bluetooth/bt-permissions. That doesn't seem to be the case anymore.

farfromrefug, 2023-12-11:
@CatchABus indeed it is now required. you can ask for it with @nativescript-community/perms

CatchABus, 2023-12-11:
> @CatchABus indeed it is now required. you can ask for it with @nativescript-community/perms

That's what I eventually did. I think we should remove any kind of permission logic in plugin (location) and update documentation to instruct users to install perms as well.

farfromrefug, 2023-12-11:
@CatchABus I feel exactly the same. never did it because it would break current usage. but I guess we could do a major update without permissions
one note though GPS state requires permission to work. same  for ble I think. 
not sure if it is an issue or not