nativescript-community/ble #279: Passkey support

ray73864, 2024-07-24:
Any chance of passkey support?

I have this working with a Zebra ZQ521 that was sent to us by a company we buy our CipherLab devices from, but it requires me to pair to the printer first via the Android Settings app.

Unfortunately we don't give our warehouse staff or store staff access to the Settings app, the CipherLab devices run in a locked down mode, which means connection to the printer via BLE would be required to occur inside the app itself.

Once initially paired though, I have it connecting, getting the list of services, selecting the correct service and characteristic, then writing to the characteristic (Chunking the data up into bite sized pieces because the labels I want to send aren't simple ones).

That all works perfectly, happens nice and quickly too, I would just love to get the whole procedure done inside the app.

Discovery, Pairing, Connecting, Writing, and Disconnecting.

farfromrefug, 2024-07-24:
@ray73864 bounding is not really part of BLE but more part of Bluetooth classic. 
So not even sure it would br part of this plugin. 
That being said it is obviously doable. However I wont do it anytime soon as I dont need it in any of my projects.
PRs are obviously welcome though.

ray73864, 2024-07-24:
Which is interesting, as I'm connecting to the printer via BLE.

What I found is that if I don't have the printer paired at all to the CipherLab device, and I use my app to connect to it via ble, it says it connected, but when I try and write to the printer using the correct characteristic, nothing happens.

If however I pair to the printer first, I can connect using the ble plugin, then send a write command, and it spits out a label just fine.

So I just 'forgot' the printer on the device, then went into my app, scanned a product barcode, then tapped on 'print', and it errors out with:

```
{
    "method": "writeCharacteristic",
    "arguments": {
        "peripheralUUID": "48:A4:93:61:32:49",
        "serviceUUID": "38eb4a80-c570-11e3-9507-0002a5d5c51b",
        "characteristicUUID": "38eb4a82-c570-11e3-9507-0002a5d5c51b",
        "value": "\n      ^XA\n      ^CF",
        "encoding": "UTF-8"
    }
}
```

The full value is:

```
      ^XA
      ^CF0,100,90
      ^FO17,16

      ^FX Draw box around label
      ^GB720,960,8^FS
      ^FWB
    
        ^FT150,940
        ^FDTEST^FS
      ^XZ
```

I have my app 'chunking' the string up into bite sized pieces because some of the things being printed will be too much for the ble protocol.

farfromrefug, 2024-07-24:
@ray73864 one of the principle of the BLE is to allow to connect to a device without any pairing. I dont know why your device requires the the bonding. But i assure you it is not needed in BLE. I have done it with many BLE devices from many different manufacturers.
It must be something related to your device / usecase.
Now i understand it is not possible for you do a PR, understand also that i maintain most of community plugins on my free time and without being paid. So mostly no new big features come without me needing it in one of my paid projects.

ray73864, 2024-07-24:
Oh I understand that with regards to the development of the plugin.

As for the bluetooth side, I'm not sure.

It's a Zebra ZQ521 RFID Mobile Printer which according to the specs supports Bluetooth 4.1 (Classic and BLE) with 'Security Models 1-4' and 'Key Bonding'.

The CipherLab device is a CipherLab RS35 with support for Bluetooth Class I, V4.2 dual mode, V2.1 with Enhanced Data Rate (EDR).