nativescript-community/ble #280: Service Not Found Error when Writing Commands to ESP32 in NativeScript BLE App

luigi7up, 2024-10-20:
I’m currently building a NativeScript app that uses BLE to connect to an ESP32 microcontroller. The microcontroller initializes a BLE server with a name, a service UUID, and characteristic UUIDs, as shown in the code snippet below. My NativeScript app successfully connects to the BLE service, but when I attempt to write commands to the characteristic, I consistently receive a `service_not_found` error.

```
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
#include <ESP32Servo.h>
#include <string>

#define SERVICE_UUID "3e64e8c5-6b16-4b5d-b244-5bfa29c64aa6"
#define CHARACTERISTIC_UUID "c18c2e14-0b84-4b1e-8c9e-2195c62ee5e2"

// Some more code...

// BLE setup
BLEDevice::init("MyESP32Locker");
Serial.println("Initializing MyESP32Locker");
BLEServer *pServer = BLEDevice::createServer();
pServer->setCallbacks(new MyServerCallbacks());

// Create BLE service
BLEService *pService = pServer->createService(SERVICE_UUID);

// Create a BLE characteristic
pCharacteristic = pService->createCharacteristic(
    CHARACTERISTIC_UUID,
    BLECharacteristic::PROPERTY_READ |
    BLECharacteristic::PROPERTY_WRITE
);

pCharacteristic->setCallbacks(new MyCallbacks());

// Start the service
Serial.println("Starting BLE service...");
pService->start();
Serial.println("BLE service started!");

// Start advertising
BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
pAdvertising->addServiceUUID(SERVICE_UUID);
pAdvertising->setScanResponse(true);
pAdvertising->setMinPreferred(0x06);  // Functions that help with iPhone connections
pAdvertising->setMinPreferred(0x12);

Serial.println("Starting BLE advertising...");
BLEDevice::startAdvertising();
Serial.println("BLE advertising started!");
Serial.println("Waiting for a client connection to notify...");
```

Native script code:

```
const Bluetooth = require("@nativescript-community/ble").Bluetooth;
const { fromObject } = require("@nativescript/core");
const createViewModel = require("@/view-models/loading-view-model").createViewModel;
const Permissions = require("nativescript-permissions");

const SERVICE_UUID = "3e64e8c5-6b16-4b5d-b244-5bfa29c64aa6"; // Service UUID
const CHARACTERISTIC_UUID = "c18c2e14-0b84-4b1e-8c9e-2195c62ee5e2"; // Characteristic UUID

async function sendCommand(command) {
    try {
        if (connectedPeripheral) {
            console.log("Sending command:");
            console.log("connectedPeripheral.UUID " + connectedPeripheral.UUID);
            console.log("SERVICE_UUID " + SERVICE_UUID);
            console.log("CHARACTERISTIC_UUID " + CHARACTERISTIC_UUID);
            console.log("[command] " + command);
            console.log("[command as Uint8Array] " + new Uint8Array([command]));

            await bluetooth.write({
                peripheralUUID: connectedPeripheral.UUID,  // Use peripheralUUID instead of UUID
                serviceUUID: SERVICE_UUID,
                characteristicUUID: CHARACTERISTIC_UUID,
                value: new Uint8Array([command]) // Using Uint8Array for command
            });
            console.log(`Command ${command} sent to Locker.`);
        } else {
            console.warn("No device connected.");
        }
    } catch (error) {
        console.error("Error sending command: ", error);
    }
}
```

Error I receive:

```
Sending command:
Error sending command: [BluetoothError]:service_not_found,arguments: {"peripheralUUID":"08:B6:1F:28:62:86","serviceUUID":"180e","characteristicUUID":"c18c2e14-0b84-4b1e-8c9e-2195c62ee5e2","value":[0]}
```

Additional Notes:

- I have verified that the characteristic and service UUIDs are correct.
- I have used the nRF Connect application on my Android phone, and it can successfully communicate with my ESP32, sending commands to the device's service with that UUID

[image]

I would greatly appreciate any help or suggestions on how to resolve this issue. Thank you in advance!

farfromrefug, 2024-10-20:
@luigi7up you need to discover services first. This is not done automatically (or can using a connect parameter). Without this services cant be used

luigi7up, 2024-10-20:
@farfromrefug hi, thank you for your reply... I¡m sorry for not providing the whole code where you can see that I'm calling the `bluetooth.startScanning({})` method

I'll update my question above

farfromrefug, 2024-10-20:
@luigi7up i am.not talking about scanning. I am talking about discovering services and characteristics. It is mandatory. 
Look at the typings for how to do it either manually or through the connect method

luigi7up, 2024-10-22:
@farfromrefug Hi! Thanks a lot for your help. It wasn’t immediately clear to me that even though I knew my `SERVICE UUID`, I couldn’t send commands to it without first discovering it during the `connect()`. Following your guidance, I was able to solve the issue by adding the` autoDiscoverAll: true` parameter to the `bluetooth.connect({})` call.

One thing I found a bit confusing is the lack of mention of service discovery in the README. If it’s necessary to discover services before interacting with them, shouldn’t the documentation include this process?

I’d be happy to contribute by opening a PR to improve this, but I want to fully understand it before doing so.

```
await bluetooth.connect({
            UUID: peripheral.UUID,
            autoDiscoverAll: true, // this is the line I added
            onConnected: (peripheral) => {
                peripheral.services.forEach(function (service) {
                    console.log("service found: " + JSON.stringify(service));
                });
                
            },
            onDisconnected: () => {
                //console.log("Disconnected successfuly to "+ peripheral.name);            
                //...
            }
        });
```
Now, the peripheral object holds services object and now you can use `bluetooth.write({})`

```
await bluetooth.write({
                peripheralUUID: connectedPeripheral.UUID,  // Use peripheralUUID instead of UUID
                serviceUUID: MY_SERVICE_UUID,
                characteristicUUID: CHARACTERISTIC_SERVO_COMMANDS_UUID,
                value: [command]
            });
```

farfromrefug, 2024-10-22:
@luigi7up you are right not everything is in the README. Please open A PR if you want to update it. This is the file to modify https://github.com/nativescript-community/ble/blob/master/packages/ble/blueprint.md.
Keep in mind that i want users to actually read platform native docs. BLE is very specific to platforms and i dont think you can really start working with it using the N plugin if you dont know how it works on the native side.
Thanks!

luigi7up, 2024-10-23:
Hey! Sure. Would you mind pointing me in the direction of the platform native docs you had in mind. I assume you mean, for example, the docs for Android/BLE https://developer.android.com/develop/connectivity/bluetooth/ble/ble-overview and the same for iOS ?

luigi7up, 2024-10-23:
Hey! Sure, I’ll definitely take a look at the file and work on a PR to update the README. I agree that having a solid understanding of the platform-native BLE docs is important. Just to clarify, are you referring to docs like [Android BLE](https://developer.android.com/develop/connectivity/bluetooth/ble/ble-overview) and the iOS equivalents? If so, I can see how they’re essential for learning the core concepts, but the code differs quite a bit from working with this plugin.

I think adding some references or context in the README would still be valuable to bridge that gap for users who might not have much experience with native BLE development. Thanks for the guidance!

farfromrefug, 2024-10-23:
@luigi7up maybe this one is better https://developer.android.com/develop/connectivity/bluetooth/ble/connect-gatt-server
But yes those are the docs i am referring too